lukavidic/ColorSorter_B #19: Issue #19 : Projektovanje pcb-a

natasamiljevic, 2024-12-23:
Po završenom projektovanju električne šeme potrebno je projektovati štampanu pločicu. Projektovanje štampane pločice podrazumijeva i pronalazak svih odgovarajućih footprinta za komponente koje će se koristiti (mikrokontroler, wifi modul, naponski regulator, eksterni oscilator), te odabir odgovarajućih dimenzija za smd otpornike i kondenzatore i odabir konektora.

natasamiljevic, 2024-12-23:
Težina zadatka na skali od 1 do 10 je 8.

Chicha13, 2024-12-25:
Ocjena težine (1-10):

8

rankomirosav, 2024-12-25:
Tezina zadatka je 8.

natasamiljevic, 2025-02-16:
Završeno. Potrebno vrijeme je 5 dana.
(uključujući vrijeme potrebno za pretragu odgovarajućih komponenata, procjenu budžeta i uklapanje svega što je ovaj task uključivao - ne samo samog projektovanja u KiCad alatu)

closed 2025-02-16 by natasamiljevic